Cancro29/NETHERCAP #1: Request for English Language Implementation. ( Especially for Esp32 )

MDRafi08642, 2023-09-18:
I want to express my deep appreciation for your significant contributions to this **_ESP32_** compatible project.
To further enhance its authenticity and accessibility, I kindly request the implementation of the English language on project pages(**_ESP32_**), as depicted in the attached screenshots.
Your support in this endeavor would be highly valuable. 👍

[image] [image] [image]

Cancro29, 2023-09-18:
> <https://user-images.githubusercontent.com/81744420/268538728-119678e8-c1bc-4f17-a46a-622924a8063b.png> [image:
> image]
> <https://user-images.githubusercontent.com/81744420/268538766-24b0c9ba-be02-4fe5-bba2-d964e848f920.png> [image:

Thank you for using Nethercap. English translation will be available on the
next release.

MDRafi08642, 2023-09-28:
> Thank you for using Nethercap. English translation will be available on the next release.
> […](#)

Any update for ESP32 ?

Cancro29, 2023-10-01:
Hello. Sorry for the late reply. The firmware is already done but i can't
get the flash backup due to esptool refuses working. I will send you the
compiled binary, this firmware can only be installed using NETHERCAP file
manager. Access 192.168.4.1 on your browser, go to file manager, upload the
file and make sure the file name ends with *.bin, then press flash button

MDRafi08642, 2023-10-01:
> Hello. Sorry for the late reply. The firmware is already done but i can't get the flash backup due to esptool refuses working. I will send you the compiled binary, this firmware can only be [...]

Ok, I will try.

JackJones58, 2024-02-06:
Can you please send me the english version?

closed 2024-02-07 by Cancro29