Cancro29/NETHERCAP #2: quanto tempo demorará a nove versão para esp32?

oikaka1, 2023-10-08:
comecei a usar o NETHERCAP recentemente e estou muito satisfeito, mas a ideia de ter funções novas me deixa muito mais animado.
então realizo a seguinte pergunta: quanto tempo até que saia a nova versão do NETHERCAP

Cancro29, 2023-10-08:
Olá, obrigado por usar o Nethercap. A versão 2.2.1 do ESP8266 chegará em breve com algumas melhorias. Quanto ao ESP32, atualmente estou desenvolvendo para a placa M5StickC Plus, então ainda precisa de algum tempo para o lançamento propriamente dito. Se você tiver alguma solicitação de recurso, não hesite em me dizer.

oikaka1, 2023-10-08:
Obrigado pela resposta rápida, espero que saia o quanto antes a nova versão. 
já que está me dando a oportunidade de pedir recursos, eu gostaria de pedir que implemente as funções de: analisar a voltagem de entrada, a opção de um ataque de beacon, correção no eviltwin, a customização do html no eviltwin, a possibilidade de esconder a rede master, quem sabe mais pra frente um suporte ao 5ghz oferecido pela placa esp32-C5, a capacidade de identificar outros ESP com o NETHERCAP via bluetooth ou wifi mesmo, por enquanto são somentes essas minha solicitações.
acho que até exagerei um pouco, mas são apenas alguns pequenos detalhes que sinto falta.
obrigado por todo seu esforço ao criar este software.

Cancro29, 2023-10-08:
Obrigado pela resposta. Explicarei suas solicitações de recursos:

1. análise da tensão e temperatura de entrada: só é possível no M5CStick Plus, pois o ESP32 Dev Kit não possui os componentes necessários.

2. ataque de beacon: já planejado.

3. correção em eviltwin: não sei o que deve ser corrigido, talvez o idioma?

4. customização do html no eviltwin: Disponível na versão ESP8266, planejada para ser implementada no ESP32.

5. a possibilidade de esconder a rede master: Disponível na versão ESP8266, planejada para ser implementada no ESP32.

6. suporte para 5ghz oferecido pela placa esp32-C5: Minha primeira prioridade quando a placa estiver disponível.

7. capacidade de identificar outros ESPs com NETHERCAP via bluetooth ou wifi: possível, mas provavelmente não acontecerá em breve.

Não posso dizer quando esses recursos estarão disponíveis, pois tenho tempo limitado para desenvolvê-los, mas você verifica minha página no Github regularmente para atualizações.

closed 2024-02-07 by Cancro29