Cancro29/NETHERCAP #3: This code is great but...

JackJones58, 2024-02-06:
Please add an english version other than it, it is amazing. Thank you!

Cancro29, 2024-02-06:
English version coming soon!

Cancro29, 2024-02-06:
Hello, English version for ESP-32 have been released. You can download it in Releases or builds directory

JackJones58, 2024-02-06:
Thank you! Do i update using the file manager or should I just reflash it, Thanks!

Cancro29, 2024-02-06:
You have to update it using file manager. The instruction is included on Releases page

JackJones58, 2024-02-06:
Works like a charm but there is a very minor bug here. 
[image]

Cancro29, 2024-02-06:
I have updated the file

JackJones58, 2024-02-06:
Yes, I tried again unfortunately the problem is still there, But everything else is perfectly fine in english.

Cancro29, 2024-02-06:
Make sure you download the file from Releases, i haven't upload the fixed file in builds directory

JackJones58, 2024-02-06:
Ohh, I wasn't downloading it from there! I wil lt now.

closed 2024-02-07 by Cancro29