Cancro29/NETHERCAP #7: Can we receive attack update on Telegram?

abdul-nnd, 2024-02-18:
There is an entry of Telegram token in config.json file. What is it? Is it working? Is it possible to receive attack update on Telegram?

Cancro29, 2024-02-18:
Yes it is possible, even remote control through telegram works too. But i
decided to disable it because it causes performance issue.

abdul-nnd, 2024-02-18:
It's a very good idea to remotely control our device. We request you, Please remove such performance issues and implement it. If possible use different services other than Telegram such as Firebase or any other.

Cancro29, 2024-02-18:
It is not possible to optimize it. Either the telegram bot will respond for
a very long time(can takes minutes) OR deauth and captive portal becoming
ineffective. Besides, that feature requires another Wi-Fi hotspot for
internet connection.

abdul-nnd, 2024-02-18:
You can use Firebase for realtime update. Another hotspot or internet connection is not a problem, we can manage it.

Cancro29, 2024-02-19:
I don't want to add another libraries

closed 2024-02-19 by Cancro29