Cancro29/NETHERCAP #9: ESP 32 verson is not working properly

Pokemonarmy360, 2024-03-22:
ESP 8266 VERSON IS WORKING GREAT I THING PLS MAKE THE PAGE LANGUAGE IN ENGLISH

MY ESP 32 IS NOT BOOTING WITH THE V6. BUT WORKING WITH V5 AND IN THE V5 U CANT GET TO  CHOOSE WHAT EVIL TWIN HTML YOU WANT TO SHOW.

ESP 8266 VERSONS ARE FINE

Cancro29, 2024-03-22:
You are supposed to update to V0.6 using OTA update in file manager. Make
sure there is enough free space before updating

Cancro29, 2024-03-22:
On version 0.7, HTML menu will be available. It will be released soon.

Pokemonarmy360, 2024-03-23:
This is reply to the ota update message

So tomorrow night I actually first discovered your post and was very much excited that now I can finally get a proper working ESP 32 jammer and evil twin

I am using a ESP 32 you with 3Db antenna, and in my set up version 5 is very much like you and version 6 is not even installing can you show me how I can install the version 6 because I have tried the method you have given in your get have to upload the file to ESP and then flash it directly but that is not even happening because when I have started flashing my ASP gets hang and I have to disconnect it from battery and reconnected again to work properly, and yes I have tried this right way when the 5th version installed without even installing the html's in the ESP I tried to update that but it is not happening so right now can you help me why it is not updating.......

Pokemonarmy360, 2024-03-23:
Reply to the version 7

Do the version 7 will be good like the nodemcu and d1 mini one?

if yes then I am willing to wait you don't have to tell me how to fix the previous issue with the update and working not properly

ESP8266 was so good i am not having any problem
That's why I am asking
Also the reason is I am making a project and if you can sure me that the version 7 will be as good as the ESP 8266 then I can save some money on my personal project because I was planning to get two new ESP 8266 and this ESP 32 you which I have already bought to make a proper web based Wi-Fi hacking set up with handshake capturing and evil twin and spacehuhn deauthor

Pokemonarmy360, 2024-03-23:
Also one thing I can change the HTML language on my own right in vs code?

Cancro29, 2024-03-23:
Thank you for your feedback on ESP32 version of NETHERCAP, it really motivates me to improve it further. 

I will try replicate how you install 0.6 version. Starting with installing 0.5 version and then update it to version 0.6 using file manager

ESP32 version is supposed to be better and faster than ESP8266 version. I just need to adapt with the board because the code will be a bit different, and compiling and testing the firmware takes longer than usual. My priority is to make the features available like ESP8266 version and then add more features like handshake capture, sniffing, and probably bluetooth stuff. 

About changing HTML language, you can modify it using any text editor.

Pokemonarmy360, 2024-03-23:
Will that the spirit! I hope you can do all of that in esp 32 but I guess in further future you are going to need more storage in ESP 32, like a SD card reader for more lines of code because when it is compressing in the v5 only a little bit space is left so i guess if you can you have to connect a SD card reader

Also can i help you by making you other website login page html linke Netflix etc, and also changing the existing ones into language like english and others

Cancro29, 2024-03-23:
Memory space is obviously an issue on ESP32 version, but i think attaching SD card is overkill. NETHERCAP is meant to be installed on bare minimum hardware, and not all boards come with SD card slot. 

I've been planning on modifying partition table to squeeze out more free space and adding direct OTA update so there's no need to upload it on file manager. I also made flashing easier for Windows PC user by providing executable flasher (go check it out)

I appreciate your help to provide HTML pages. You can send me a link for said HTML pages.

closed 2024-03-28 by Cancro29