Cancro29/NETHERCAP #10: [ESP8266] Custom Rogue AP

Miku-33, 2024-03-24:
Hello,

First, I would like to thanks you a lot for this work !

I would like to add my own rogue AP with multiple html pages and some assets for css and js code (it run well on ESP8266 with basic captive portal program).

The first page (renamed /rogue2) will ask to choose the account for log in, for example the code redirect to /google-login.html with a href button. Then the form in it will submit as your facebook exemple with argument named email and password then redirect to "/redirect" (action form) which is an html page with an error info and a button to get back to menu /rogue2.

I put all my pages in "root" (?) with the file manager (same directory as rogue2 file). When I run the Rogue AP, it loads well the first page but I can't access my others html file with href buttons (like there is an handleroot who redirect to rogue2 each time buttons are clicked)

Maybe the file manager doesn't put files in "root" directory ? I think because I can't find your '502 error' page here..! So could you give me the entire directory name to update my html pages ?

Thanks a lot, and sorry for my french-english talk ;)

Miku-33, 2024-03-24:
Hi, I would also tell you that the Evil Twin portal don't show up on laptop (no problem with the main NETHERCAP's or Rogue AP's captive portals). Maybe due to a difference btwn DNS settings ?

Cancro29, 2024-03-24:
Hello Miku-33, thank you for using NETHERCAP. I will explain how Rogue AP works:

When Rogue AP is initiated, it will lock the path of selected HTML file (in this case /rogue2.html) and it will be placed in main address (192.168.4.1/), replacing main admin page so the captive portal will correctly redirect to said HTML. Your HTML page contains "/google-login.html" and "/redirect", which NETHERCAP does not understand because there are no functions to handle that address despite it being correctly placed in root directory. 

Every address like "/monitor", "/settings", "/fsbrowser" etc are predefined with their own function handler so it can send the correct page to the user. Any address other than that will not work because NETHERCAP does not know how to handle that address.

There is two way to fix this issue:
1. You have to contain all of your HTML elements into one file, including CSS and JS, and fake that redirecting stuff. Lets say there is two pages, one is "choose account" page and the other is "login with google" page. Initially, "login with google" page is hidden. After "login with google" button is pressed, hide "choose account" page and then show "login with google" page.
2. I have to modify the source code. That involves scanning all HTML file and then create a handler for each HTML. Theoretically, i can do this easily, but you have to be careful as you are creating your own Rogue AP workflow, you need to carefully redirect into correct address. Keep in mind that NETHERCAP only takes 2 arguments for Rogue AP, "email" and "password". Make sure you named the form ID correctly.

About '502 error' page, it was hard coded. It does not exist as file, just string.

About Evil-Twin portal not working on laptop, that's an unusual behavior but i expected that. I used same DNS settings on all captive portal, and it does work on my laptop. I need to test it further using different PCs.

Anyway, thank you so much for your feedback, it definitely sparked an idea to add more functionality to this firmware.

Miku-33, 2024-03-24:
Thanks for your reply !

I'll try to merge files int one html page, but it seems a bit complicated (with my too little knowledge in html/css programming).

Is there a way for me to try to modify the source code with these handlers ? (Not necessary scanning files because in fact their names are known, right ?).
If you're interested, I can share you theses files).

Also, I've detected a little bug on the "error 502" html file after submitting, there are quotes marks " before and after on the rendering.

Cancro29, 2024-03-25:
Unfortunately, the code is closed souce. No one except me can modify it. About the bug, it will be fixed on later release.

Miku-33, 2024-03-25:
Ok, So I will wait for a possible update. Thanks!

Cancro29, 2024-03-28:
> Ok, So I will wait for a possible update. Thanks!

Hello, can you provide me a sample of your HTML, CSS, and JS file. I've managed to make it work and I need your files to make sure it works properly.

Miku-33, 2024-03-29:
Hi,

You can download files here (WeTransfer) : https://we.tl/t-NGZpv8twHW

Feel free to ask me for more informations if you're in need.

CSS don't work on a laptop when you open html file directely, but it's OK trough a server.

Cancro29, 2024-03-29:
Hello Miku-33, I've tried your project files and i'm impressed on how complex and well-made the project is. I've managed to make it work with a few notes: 

1. The project you just sent to me uses directory on it's filesystem, which NETHERCAP does not support. However, I found a workaround by adjusting the path of HTML elements so it points to root directory. 
2. I can provide you a SPI Flash backup of my ESP8266 so you can just flash it on yours and it will be ready-to-use. Just tell me which board you are using and I will send it through email. 
3. Looks like I need to add directory support on NETHERCAP😄. Putting all files in root directory makes them look messy.

Miku-33, 2024-03-30:
To be honest, they aren't totally mine, I found somes base files on others repositories and then edited them to be as I want to ! 

That's a good new! I'm using an ESP8266 WEMOS D1 Mini 4 MB. You can share me a link to DL it ? 

Right, there are a lot of files..😄

Cancro29, 2024-03-30:
Sure! , but I can't share using public link as I don't want to make it public for now. You can send me an email so I can send it directly to you. 

Just curious,if that project isn't fully yours, where did you get it from?

Miku-33, 2024-04-02:
For example, I was inspired by the following projects (thanks for their work !) : 

https://github.com/FlippieHacks/FlipperZeroEuropeanPortals/tree/main?tab=readme-ov-file
https://github.com/SpacehuhnTech/esp8266_deauther
https://github.com/trumtuong/ESP8266-WifiPhisher
https://github.com/OxMarco/RogueCaptivePortal/tree/master
https://github.com/FlippieHacks/EvilPortalGenerator?tab=readme-ov-file
https://github.com/TasosY2K/ESP8266-Phising?tab=readme-ov-file

Cancro29, 2024-04-05:
Thank you for the links!. I have sent the file to you.

closed 2024-04-05 by Cancro29