Cancro29/NETHERCAP #11: Display and switch pin layout?

Pokemonarmy360, 2024-03-28:
Hi this is me again I am actually working on the HTML and this time I actually want to know the pin layout for display and switches are same with the spacehuhn deauther? Or it has other pin layout? I am making a pocket version of it that's why I am asking....... 
[image]

Pokemonarmy360, 2024-03-28:
[image]
Also making a big one for the ESP 32 [your version] and spacehuhn deauthor

Cancro29, 2024-03-28:
Hello, the wiring of ESP8266 OLED looked like this :
VCC  -> 5V
GND -> G
SCL  ->  D1
SDA  -> D2

And for ESP32:
VDD  -> 3V3
GND -> GND
SCK  -> D22 
SDA  -> D21

Pokemonarmy360, 2024-03-28:
Is there any press switch pinout for going through UI in the display?

Cancro29, 2024-03-28:
On ESP8266, you can attach a button on pin D7.

Pokemonarmy360, 2024-03-29:
Is there only one button in ESP 8266? Or there is other two buttons?

Cancro29, 2024-03-29:
Only one button is supported, and the operation involves pressing and press&hold

Pokemonarmy360, 2024-03-29:
Oh ok thx

closed 2024-04-06 by Cancro29