Cancro29/NETHERCAP #12: Custom pages for Connecting/Retrying/Password Failed

krisss117, 2024-05-13:
Hi, it is amazing project. 

If it is possible it will be amazing to set pages for: Connecting/Retrying/Password Failed pages. 

I think You can use File Manager and set uploaded page as Connecting/Retrying/Password Failed :)

krisss117, 2024-05-15:
I think it will be ok. Information in local language are more trusted.

Cancro29, 2024-05-18:
Hello, I have released version 2.5 which support custom language on default captive portal and reconnecting page.

krisss117, 2024-05-18:
Amazing! Work perfect :) 

Please add one more variable to change "Password" :) 

PS. Thanks for fast implement changes

Cancro29, 2024-05-18:
I'm sorry, I don't understand that. Can you give me further explanation?

krisss117, 2024-05-20:
[image]

Cancro29, 2024-05-21:
Ah yes, i forgot to add that. Thanks for reminding me

krisss117, 2024-05-23:
Hi, text box cut words - show only first words...
[image]
[image]

Cancro29, 2024-05-23:
Thanks for pointing it out. It will be fixed in later release

closed 2024-05-29 by Cancro29