Cancro29/NETHERCAP #15: Send logs by email after reboot

krisss117, 2024-05-15:
Hi, sorry for spam, but this project is amazing :) i have few ideas, and want to share it with you :) 

It is possible to add 
- WiFi networks (SSID + Password) to settings - textbox 
##############
SSID;PASS
SSID;PASS
SSID;PASS
SSID;PASS
SSID;PASS
##############
- SMTP credentials to settings
- checkbox "Send LOG to Email"
- checkbox "Continue attack after send logs"
- inputbox "Email:"

When something new show in log -> then try to connect to WiFi AP (foreatch list of WiFi networks) when connection successful then send logs and if "Continue attack after send logs" checked then continue attack. 

What do you think, it is possible?

Cancro29, 2024-05-15:
No worries! I'm happy to see feedbacks from people who used my firmware.

1. It may be possible to WifiMulti to manage connection to internet source.
2. SMTP logging was already exist before, but I disabled it because lack of RAM and CPU.
3. checkbox "Continue attack after send logs". I assume this is for RogueAP?.
4. inputbox "Email:". I don't understand this.

I may make this possible, but probably only for ESP32.

krisss117, 2024-05-15:
3. Yes
4. Destination email - email to send logs

Ok, can't wait when ESP32 support all functions 😁

closed 2024-09-18 by Cancro29