Cancro29/NETHERCAP #16: PMKID & WPA/WPA2 Handshake capture

krisss117, 2024-05-15:
In the future, you can add features such as:

- PMKID capture
- WPA/WPA2 handshake capture and parsing
- Deauthentication attacks using various methods
- Formatting captured traffic into PCAP format
- Parsing captured handshakes into HCCAPX file ready to be cracked by Hashcat
- Passive handshake sniffing

From https://github.com/risinek/esp32-wifi-penetration-tool

Cancro29, 2024-05-15:
I've considered making this feature available, but I'm having hard time understanding risinek's code. Probably won't coming soon enough.

TheAjaykrishnanR, 2024-06-05:
is it even possible ? I have read in an article that esp 8266 truncation does not allow it to capture the 4 way handshake : https://www.varonis.com/blog/hacking-wi-fi-with-the-esp8266

Cancro29, 2024-06-08:
Yes, it is only possible on ESP32. He meant implementing handshake captire on ESP32 version of NETHERCAP, not ESP8266 version

closed 2024-09-18 by Cancro29