Cancro29/NETHERCAP #21: m5stick plus no longer working?

banned747, 2024-06-08:
ive got an m5stick c plus and cannot figure out how to install it. ive tried using random bootloader and partitions but nothing is working. can you point me to working ones? thank you.

Cancro29, 2024-06-08:
You can download it from here : https://drive.google.com/file/d/17dXTXMxgzy0Dm0D5MiYdLRL1RMJkDbOh/view?usp=sharing. This is a full SPI flash backup so you can just flash it with 0x0 offset. Keep it mind that this build is still in early stage of development so you might be a bit buggy sometimes.

banned747, 2024-06-09:
thanks for the reply. i have uploaded it to my m5stick but it does not boot. it just stays black screen.

banned747, 2024-06-09:
image guide is a great addition to btw, cheers.

Cancro29, 2024-06-09:
Welp, turns out I'm unable to flash it to my M5Stick too. I need some time to figure it out

Cancro29, 2024-06-16:
Hello, I've fixed the M5StickC Plus build. You can check it here https://github.com/Cancro29/NETHERCAP/tree/main/ArchivedBuilds/M5StickCPlus

closed 2024-07-28 by Cancro29