Cancro29/NETHERCAP #22: Rogue AP for esp32 and a readme

j-e-e-v-a-n, 2024-06-10:
Please fix Rogue AP for esp32 can you add a readme on how to use every feature and add a feature too show login credentials that has entered in rogue AP when connected by admin

Cancro29, 2024-06-11:
Rogue AP will be coming soon, but PCAP capture won't come anytime soon

j-e-e-v-a-n, 2024-06-14:
what about the beacon spammer

Cancro29, 2024-06-14:
Beacon spammer is in lower priority list as i want to implement it first in ESP8266. CLI version? Maybe. BLE spoofer, not interested. Touch + buttons, why? Just use your smartphone.

In other words,I have limited time, limited coding skill, and there are too many variants of esp32 boards and its modules. Beside that, the SDK has a flaw that deter me from developing it. I'd rather develop the firmware for bare boards and control it with web server, as it is easier and cheaper for me and people.

Cancro29, 2024-06-14:
I want to implement those features too, but it will take a very long time to code and test for each board version, which I couldn't. My current aim is to make esp32 version controllable through esp8266, so there is possibility to capture handshake using esp8266 with help from esp32, and possibly deauth 5 GHz wifi after esp32c5 gets released

j-e-e-v-a-n, 2024-06-14:
Or you can make a team of devops who has the same aim as you

You can find a lot who have skill,time and  can contribute in some tg groups

Cancro29, 2024-06-15:
RogueAP is just a fake captive portal that asks for username and password, mostly mimic social media login page. It is still being worked for ESP32, but it already works on ESP8266

j-e-e-v-a-n, 2024-06-15:
@Cancro29 ok thanks

BgLuc88, 2024-07-21:
Tôi thực sự mong chờ tính năng tấn công bluetooth ❤️

closed 2024-09-30 by Cancro29