Cancro29/NETHERCAP #23: Nodemcu Version In oled The Language Is Indonathioan

ridwan4200, 2024-06-12:
How To Set The Language To English In evil Twin on oled version? 
[image]

Cancro29, 2024-06-12:
There are language option in settings

ridwan4200, 2024-06-13:
but how i can get in setting cause im using oled display and button

Cancro29, 2024-06-13:
Use wifi to connect to SoftAP and access 192.168.4.1 using browser

ridwan4200, 2024-06-13:
if i change the language with browser after that if i do the evil twin in oled display without browser will it solve ?

Cancro29, 2024-06-14:
Why don't you just try it? You can find the answer faster than waiting for my answer

ridwan4200, 2024-06-14:
> Why don't you just try it? You can find the answer faster than waiting for my answer

did you have any other social media of your where i can ask you and you can answer faster?

Cancro29, 2024-06-14:
You can contact me via email at nethercap.dev@gmail.com

ridwan4200, 2024-06-16:
It worked. Thanks For You Replies And I Really Love This Project. You Can Remove This Issue Now

closed 2024-09-30 by Cancro29