Cancro29/NETHERCAP #26: How to connect navigation key?

BgLuc88, 2024-07-19:
Can I connect the navigation keys up and down to confirm ok or return? esp32, esp32c3, esp8266, or wemos d1 pro? Which models can connect navigation keys?  I found jump instructions D3 D5 D6 and D7. 1 key will have 2 connecting wires, for example the first wire connects to D3, so will the 2nd wire connect to GND or Vcc? Thanks you very much ❤️

Cancro29, 2024-07-20:
Hello, thanks for using NETHERCAP. Unfortunately, the button control scheme is hard coded and cannot be changed. It also only exists on ESP8266 build. For buttons, one of the pin is connected to 3.3v pin.

I appreciate that you are trying it on ESP32. To me, ESP32 boards have a lot of processing power, thats why it is fast at verifying passwords, but it has worse web server performance with random disconnect and overall slower server response. I think it is not worth updating it. I suggest to use ESP8266 because it is the most usable and stable. Besides, i have a plan to integrate ESP8266 with 5 GHz deauther, which make it perfect for deauthing device

BgLuc88, 2024-07-20:
Thank you. i will try it on wemos d1 and esp32c3 and feedback the results. one more time thank you very much

closed 2024-07-20 by Cancro29