Cancro29/NETHERCAP #32: Is the display connection available on esp32c3?

BgLuc88, 2024-08-07:
Hello. I'm back. I have prepared the necessary boards to use your software. I tested on wemos d1 mini pro with ssd1306 0.96 display (display quite little information) and esp32 c2. can i connect ssd1306 display to esp32c3? What do the connected pins look like? Thank you
[image]
[image]
[image]

Cancro29, 2024-08-07:
Hello. I only tested OLED display on ESP32 Dev Kit. Theoretically It can be connected with ESP32C3, but i need to reassign display pin, as ESP32C3 does not have GPIO 21 and GPIO 22

n0tAv1ru5, 2024-08-17:
Can you release button wiring tutorial?

closed 2024-09-30 by Cancro29