Cancro29/NETHERCAP #34: Issue login page

Maam5, 2024-08-15:
Hello dear,
Hope you are doing well. I am from Bangladesh. Recently i install nethercap wifi tool on nodemcu esp8266. But i found an problem. 

That is when i action deauthor and evail twin, normally ssid change and see fake login page.  But while that process i can't manage or login 192.168.4.1 ip address to operate system. It required disconnect power and re connect.

Can you Plz give an solution. To login page during action performing.

Cancro29, 2024-08-15:
If the captive portal does no appear after activating evil-twin, try erasing your esp8266 first using Spacehuhn's reset bin file

Maam5, 2024-08-16:
can i login as a admin panel while evil twin attak run? like other tool
have. because 192.168.4.1 i enter it redirect to captive portal. give
sollution to enter as admin user on firmware while attak running.

Cancro29, 2024-08-16:
Once evil twin is started, admin page gets replaced with evil twin page. You can go to 192.168.4.1/monitor to monitor and control it.

closed 2024-08-16 by Cancro29